Xinyuan-LilyGO/LilyGo-T-SIM7080G #146: MODEM_CATM_NBIOT: initial registration does not work

gusgonnet, 2025-05-28:
When setting up a brand new device, the initial registration does not work with:

    modem.setPreferredMode(MODEM_CATM_NBIOT);

I can force the device to register with what I know works in my area, say:

   modem.setPreferredMode(MODEM_CATM);

And once that initial registration works, I can change the code to use both modes and it works fine:

    modem.setPreferredMode(MODEM_CATM_NBIOT);

But it would be ideal that the modem decides this, since I do not know where exactly the devices will end up being deployed.

Any insights as to why this would not work JUST the first time?

Thanks!

lewisxhe, 2025-05-29:
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/90

Refer to the settings here to see if they are the same.

gusgonnet, 2025-06-02:
Band settings are the same:

```
AT+CBANDCFG?

+CBANDCFG: "CAT-M",1,2,3,4,5,8,12,13,14,18,19,20,25,26,27,28,66,85
+CBANDCFG: "NB-IOT",1,2,3,4,5,8,12,13,18,19,20,25,26,28,66,71,85

OK
```

Network Mode Preference is in **2 Automatic**:

```
AT+CNMP?

+CNMP: 2

OK
```

AT+CMNB Preferred Selection between CAT-M and NB-IoT is set to **3 CAT-M and NB-IoT**

```
AT+CMNB?

+CMNB: 3

OK

```

Any other ideas?
Do you know how the modem tries CATM vs NBIOT?
Thanks

lewisxhe, 2025-06-03:
Use the AT+CMNB command to force the network type. This command is restricted by AT+CBANDCFG. You must know your network band configuration, otherwise you will not be able to search for the network.

gusgonnet, 2025-06-03:
Ok, thanks.

From my perspective, the automatic selection does not work.

```
modem.setPreferredMode(MODEM_CATM_NBIOT);
```

I'm not sure how the modem implements this but it doesn't seem to work.
You can close this issue if you want. I will implement CATM vs NBIOT on my code.
Thanks!

closed 2025-06-17 by gusgonnet