Xinyuan-LilyGO/LilyGo-T-SIM7080G #147: sim7080_sim7500_sim7600_sim7900_sim8200 qdl v1.58 only for update.exe does not see COM10 port for firmware

Nyaincat, 2025-05-30:
https://www.youtube.com/watch?v=lGctqLDGHz4
я следую по этому гайду, дошел до момента 1:55, у меня некоторые вопросы, 

1.Порт MicroUSB должен быть включен?
2. При включении питания Type-c, я удерживаю кнопку BOOT, в диспетчере устройств плата отображается как "Устройство с последовательным интерфейсом USB (com10)".
В гайде у человека она отображается как simcom hs-usb qdloader 9008 (COM42)

Nyaincat, 2025-05-30:
[image]

hmm... When installing the driver for the SIM, he writes to me that an error has occurred

lewisxhe, 2025-05-30:
Update the firmware using MICRO-USB instead of USB-C. You must have both plugged in
When upgrading, write ATDebug firmware, first confirm the current firmware version, no need to press BOOT button. Otherwise, the modem will not open.

closed 2025-07-14 by github-actions[bot]