Xinyuan-LilyGO/LilyGo-T-SIM7080G #159: Unable to detect port for Lilygo TSIM7080G

schilliard, 2025-09-22:
I running Windows 11 and have followed directions to install driver from Lilygo-T-PCIE-Master folder in Github, but when I open the 'upgrade tool' from the same file, it does not detect the port that 7080G is connected. I have uninstalled and re-installed the driver a few times, restarted computer and I have run both the driver installation file and the upgrade tool file as an administrator. 

I have been following instructions on the linked Youtube file 'How to update the firmware of SIM7080G series products' found on the github page https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G

I think it is a driver issue, but I'm not sure how to fix it. Appreciate any help, thank you

[image]

[image]

[image]

lewisxhe, 2025-09-25:
https://drive.google.com/file/d/1-SlbQ9FUZzov96Xc74SJ375U9v1JVjma/view?usp=sharing

I put it on Google, although it doesn't say 7080, it still applies to 7080.

schilliard, 2025-09-26:
Still showing that it can't find the driver. Tried on two different computers running Windows 11

https://github.com/user-attachments/assets/b853bcc6-df3a-4426-b3c8-4312e02c4bf4

schilliard, 2025-09-26:
When I use the PCIE Master file, I get a different error:

https://github.com/user-attachments/assets/9c301d12-9193-40ba-9740-c1fdbe379f92

lewisxhe, 2025-09-27:
Sorry, the link above is wrong, the name is too confusing, sorry, please reinstall the driver.
https://drive.google.com/file/d/1YGMAH57G7QRPPg9Vrx_PJX9Z7VdwxhIF/view?usp=sharing

schilliard, 2025-09-29:
Thanks, it worked!